slowtec/tokio-modbus #270: Is it possible change trait Service to be a async trait?

gustavowd, 2024-07-30:
I'm using tokio modbus in several embedded applications. But i'm facing some hangins due to tokio mutex. The "call function" in the Service trait is not an async function. So, i can not use the tokio mutex to manage data concurrency in that function. In order to do that i'm using a second futures runtime with "futures::executor::block_on". However, sometimes the application hangs. Is it possible to change the Service trait to be a async trait in a way that the call function becames a async fn?

Thanks in advance

uklotzde, 2024-07-30:
The `Service` trait is already asynchronous by returning a `Future`.

closed 2024-07-30 by uklotzde

gustavowd, 2024-07-30:
> The `Service` trait is already asynchronous by returning a `Future`.
> 
> Reference:
> 
> https://github.com/slowtec/tokio-modbus/blob/91c9f8dd9ea3eab289c57e966ce08b7ab4f166a7/src/server/service.rs#L32

But how can i use a tokio mutex in the call function? I can't use any function with .await in that function.

flosse, 2024-07-30:
> But how can i use a tokio mutex in the call function? I can't use any function with .await in that function.

I think this is not a  `tokio-modbus` specific question, but a more general one.
Apart from that, it is difficult to answer your question without a specific code.

gustavowd, 2024-07-30:
> > But how can i use a tokio mutex in the call function? I can't use any function with .await in that function.
> 
> I think this is not a `tokio-modbus` specific question, but a more general one. Apart from that, it is difficult to answer your question without a specific code.

I believe it is a tokio-modbus question, cause if you implement a gateway for modbus RTU to modbus TCP/TLS using tokio-modbus, a tokio mutex is needed. If you use tokio mutex, you should use .await to lock the mutex in the call function of the Service trait. It is not possible to use this .await call in the same tokio runtime thread.

Code example:
`struct Service {
    data: Arc<Mutex<Models>>,
}`

`impl Service {
    fn new(init_data: Arc<Mutex<Models>>) -> Self {
        Self {
            data: init_data,
        }
    }
}`

`impl tokio_modbus::server::Service for Service {
    type Request = SlaveRequest<'static>;
    type Response = Response;
    type Exception = Exception;
    type Future = future::Ready<Result<Self::Response, Self::Exception>>;`

    `fn call(&self, req: Self::Request) -> Self::Future 
    {
        let res = match req.request.clone() {
            Request::ReadHoldingRegisters(addr, cnt) => {
                    let models = self.data.lock().await;
                    ...

`

flosse, 2024-07-30:
https://rust-lang.github.io/async-book/03_async_await/01_chapter.html

uklotzde, 2024-07-30:
The example has everything you need: https://rust-lang.github.io/async-book/03_async_await/01_chapter.html#async-move

gustavowd, 2024-07-30:
> The example has everything you need: https://rust-lang.github.io/async-book/03_async_await/01_chapter.html#async-move

I don't beleive it has, but ok. I will try to understand how. I know the basics of async/await. I solved the problem with another executor. Support for async traits is something new in Rust. And the problem of using an asynchronous mutex across multiple executors is widely discussed as something that can cause a deadlock. So this is not a trivial issue. But I will try to understand how that code can help me.

uklotzde, 2024-07-30:
```
fn move_block() -> impl Future<Output = ()> {
    async move {
    }
}
```

gustavowd, 2024-07-31:
I can't figure out how to use this approach, because Rust keeps telling me that this future won't be executed.
The best way that I found to execute a tokio Mutex inside the call function of the Service trait is:

                tokio::task::block_in_place(|| {
                    tokio::runtime::Handle::current().block_on(async {
                         // Run the async code
                    })
                  });

I can not use the "async move" cause I need to use some variables outside this async block.
The idea is something like this:

struct Service {
    data: Arc<Mutex<Models>>
}

impl Service {
    fn new(init_data: Arc<Mutex<Models>>) -> Self {
        Self {
            data: init_data
        }
    }
}

impl tokio_modbus::server::Service for Service {
    type Request = SlaveRequest<'static>;
    type Response = Response;
    type Exception = Exception;
    type Future = future::Ready<Result<Self::Response, Self::Exception>>;

    fn call(&self, req: Self::Request) -> Self::Future 
    {
        let res = match req.request.clone() {
            Request::ReadHoldingRegisters(addr, cnt) => {
                let mut start_addr = 0;
                let mut registers: Vec<u16> = Vec::new();
                let mut end_addr = 0;

                tokio::task::block_in_place(|| {
                    tokio::runtime::Handle::current().block_on(async {
                        let models = self.data.lock().await;
                        if (SUNSPEC_INIT_ADDR..SUNSPEC_INIT_ADDR + 2).contains(&addr) {
                            registers = models.id.into();
                            start_addr = SUNSPEC_INIT_ADDR;
                        }else {
                            for model in models.models.iter() {
                                if (addr >= model.start_addr) && (addr < model.end_addr) {
                                    let model_tmp = model.clone();
                                    registers = Vec::<u16>::from(model_tmp);
                                    start_addr = model.start_addr;
                                }
                            }
                        }
                        end_addr = models.models[models.models.len()-1].end_addr;
                        drop(models);
                    })
                });

                if addr >= SUNSPEC_INIT_ADDR && (addr + cnt) <= end_addr {
                    let registers = &registers[(addr-start_addr) as usize..(cnt+(addr-start_addr)) as usize];
                    Ok(Response::ReadHoldingRegisters(registers.to_vec()))
                }else {
                    Err(Exception::IllegalDataAddress)
                }
            },
            _ => Err(Exception::IllegalFunction)
        };
        future::ready(res)
    }
}

I follow the tokio documentation to implement that way.