slowtec/tokio-modbus #282: Breaking change with addition of `FunctionCode::ReportServerId` in a SemVer-compatible version

domenicquirl, 2024-09-12:
The 0.14.1 release breaks all users of version 0.14.0 by introducing the new `FunctionCode::ReportServerId`, which needs to be handled when matching against a `Request`. You seem to be aware of this already given that the change is called out as a breaking change [in the Changelog](https://github.com/slowtec/tokio-modbus/blob/main/CHANGELOG.md#v0141-2024-09-10). 

However, the change was published as a patch version, which means that `cargo` will happily update any users of version 0.14.0 to the new release, breaking their builds. Kindly, please do not release breaking changes in this way where running `cargo update` will make your users' code stop compiling. The new version should have been released as version 0.15.

I have handled the breaking change in my own code now, so I'm personally indifferent as to what to do with the version 0.14.1 release, but it might be best to yank it once a correctly versioned release is available.

uklotzde, 2024-09-12:
Valid complaint.

This was a deliberate decision, assuming that users of the client-side API should be unaffected. Probably short sighted in retrospect. See also: https://github.com/slowtec/tokio-modbus/pull/274#issuecomment-2340216427

Yanking v0.14.1 and holding back all breaking changes for a v0.15 release is probably the best option now.

@flosse ?

uklotzde, 2024-09-12:
I have yanked v0.14.1.

closed 2024-09-12 by uklotzde

domenicquirl, 2024-09-12:
Highly appreciate your understanding @uklotzde. Your comment on the PR of splitting the crate in two (or probably 3, with a base crate for all the shared stuff) sounds like it would be a good idea if the intent is to decouple changes to the client(s) from those to the server(s). As it stands right now, there is no way to do this in a SemVer compliant way with the single crate exposing both as feature flags.

You also call the Modbus server "experimental" in your comment, which surprised me since we have been using a `tokio-modbus` server in our code for around 5 years now and I didn't know anything about that. Is this documented anywhere? It's not mentioned in the crate docs at all as far as I can tell. That said, the server (in it's different iterations through the different versions of `tokio-modbus` from version 0.3 to now) has been very functional and, at least since it's possible to get slave IDs from requests and send "real" Modbus exceptions as responses, also doesn't feel like it would be particularly experimental.

uklotzde, 2024-09-12:
> Thanks for your understanding @uklotzde. Your comment on the PR of splitting the crate in two (or probably 3, with a > You also call the Modbus server "experimental" in your comment, which surprised me since ...

It was "experimental" in the past and somehow sneaked in ;) Initially, the main focus of this crate was the client-side API. But now more and more people seem to use it for implementing a server.

snejugal, 2024-09-12:
Perhaps it's worth adding `#[non_exhaustive]` for `FunctionCode` before 0.15.0 to ease such situations in the future.

uklotzde, 2024-09-12:
> Perhaps it's worth adding `#[non_exhaustive]` for `FunctionCode` before 0.15.0 to ease such situations in the future.

Alternatively we could add all predefined codes from the spec like we did for `ExceptionCode`.

uklotzde, 2024-09-14:
Follow-up: #284

PR #283 (2024-09-14): Update changelog after yanking v0.14.1
Fixes #282.