slowtec/tokio-modbus #293: rtu server receives but does not send

deid84, 2024-10-16:
Hello

I tried your lib in both tcp and rtu mode. While tcp works like a charm, I see that rtu receives and parse correctly the frame but it does not write the answer to the serial port.

I saw that also your rtu server example has this problem, is that a problem emerged with the latest update? 

thank you

uklotzde, 2024-10-16:
The `process()` loops for TCP, RTU, and RTU-over-TCP are almost identical. All send out a response if returned from `call()`.

uklotzde, 2024-10-16:
The new debug logs might provide more details: https://github.com/slowtec/tokio-modbus/pull/294

But I can't spot a bug.

deid84, 2024-10-17:
yeah I think I figured it out,

I didn't mention that i'm using an embedded device with a rs-485 port, looking at some examples I discovered that I need to manage RTS at low level, so it's not a problem related to your crate but to tokio-serial at a lower level.

Thank you for your support

closed 2024-10-17 by uklotzde