slowtec/tokio-modbus #309: Some doubts

1787611685, 2024-12-16:
When the modbus tcp slave device goes offline, how do I try to reconnect? I don't see any relevant methods. Do I need to write the timeout judgment and reconnection mechanism myself?

fxb666fdfds, 2026-11-12:
Yes—the reconnect policy belongs in the application. In the current implementation, [`tcp::connect_slave`](https://github.com/slowtec/tokio-modbus/blob/main/src/client/tcp.rs) opens a new TCP stream and constructs a new `Context`; the context does not retain an address and automatically reopen the connection.

A useful structure is an outer connect/retry loop and an inner polling loop:

1. Connect with a deadline, using `tokio::time::timeout(connect_timeout, tcp::connect_slave(address, slave)).await`.
2. Wrap each request in a deadline too. A successful TCP connection does not guarantee that a later Modbus request will receive a response.
3. On a transport/protocol error or request timeout, leave the polling loop and drop that context, then reconnect after a bounded backoff. Recreate it with the same slave/unit ID.
4. On an ordinary Modbus exception response, handle the exception separately; reconnecting will not fix an invalid register address.

For the current 0.17 API, the request result is nested, so the decision inside the polling loop looks like:

```rust
match tokio::time::timeout(
    request_timeout,
    ctx.read_holding_registers(start_address, count),
).await {
    Ok(Ok(Ok(values))) => {
        // Process values, then wait until the next poll.
    }
    Ok(Ok(Err(exception))) => {
        // Server answered: handle this Modbus exception.
    }
    Ok(Err(error)) => {
        // Transport/protocol failure: exit this polling loop.
        break;
    }
    Err(_elapsed) => {
        // Request timed out: exit and discard this context.
        break;
    }
}
```

This fragment assumes `Reader` is imported and that `ctx` is scoped inside the outer connect loop, so `break` drops the old context before reconnecting. Include shutdown/cancellation and a retry limit or capped backoff in that outer loop. In particular, avoid blindly replaying writes: after a timeout, the device may already have performed the write even though you did not receive its reply.

I checked the current connection/request implementation and nested `Result` type; I have not compiled this fragment or tested a device disconnect/reconnect here.

1787611685, 2026-11-12:
Thanks a lot for this detailed guidance!
I totally get it: tokio-modbus itself doesn’t carry built-in reconnection logic, all retry/backoff logic belongs to the application layer.
Your layered loop design (outer reconnect loop + inner polling loop) is exactly what I need.
Also the distinction between Modbus exception response and transport/timeout error is very helpful, I will separate them properly and add capped backoff, plus idempotent write protection to avoid duplicate write issues.
I’ll implement this structure and test it on our Modbus slave.

 

 
  
 
 
 
  
 match tokio::time::timeout(     request_timeout,     ctx.read_holding_registers(start_address, count), ).await {     Ok(Ok(Ok(values))) =&gt; {         // Process values, then wait until the next poll.     }     Ok(Ok(Err(exception))) =&gt; {         // Server answered: handle this Modbus exception.     }     Ok(Err(error)) =&gt; {         // Transport/protocol failure: exit this polling loop.         break;     }     Err(_elapsed) =&gt; {         // Request timed out: exit and discard this context.         break;     } }