slowtec/tokio-modbus #315: Attaching multiple slaves

huntc, 2025-02-11:
Just wondering if I'm missing something here... is it possible to have multiple slaves attached to the bus?

flosse, 2025-02-11:
This is of course possible with Modbus RTU.

huntc, 2025-02-11:
> This is of course possible with Modbus RTU.

Yes, but the API here doesn't make it obvious to me. Are you able to put me right? Thanks.

flosse, 2025-02-11:
> Yes, but the API here doesn't make it obvious to me.

https://github.com/slowtec/tokio-modbus/blob/53f3a08007bcd6a6d1d098c652140de5fdd711dd/examples/rtu-client.rs#L18

What are you missing?

huntc, 2025-02-11:
`attach_slave` consumes `port`, right? So how does one attach another slave to the same port?

flosse, 2025-02-11:
> `attach_slave` consumes `port`, right?

correct

> So how does one attach another slave to the same port?

Good point 🤔
In fact, I've never been in that situation myself.
We would need a kind of “detach” function that returns the port.

huntc, 2025-02-11:
Nice to know I've not gone completely nuts! Our situation is that we can have multiple power meters on the same bus. MODBUS is, after all, a bus topology. :-)

flosse, 2025-02-11:
> MODBUS is, after all, a bus topology. :-)

yes, at least the RTU variant 😄 

As a workaround, you can perhaps build a wrapper for the port that implements
`T: AsyncRead + AsyncWrite + Debug + Unpin + Send + 'static`

flosse, 2025-02-11:
Maybe you can also use `SerialStream::pair`: https://docs.rs/tokio-serial/latest/tokio_serial/struct.SerialStream.html#method.pair

huntc, 2025-02-11:
> > MODBUS is, after all, a bus topology. :-)
> 
> yes, at least the RTU variant 😄
> 
> As a workaround, you can perhaps build a wrapper for the port that implements `T: AsyncRead + AsyncWrite + Debug + Unpin + Send + 'static`

Yeah, I was thinking I've implementing my own port anyhow as I've got to control a tx GPIO. So I guess I could make my port be clonable...

> Maybe you can also use SerialStream::pair: https://docs.rs/tokio-serial/latest/tokio_serial/struct.SerialStream.html#method.pair

Interesting. Thanks. Any alternate thoughts on the control of the GPIO for bracketing the write then?

flosse, 2025-02-11:
> Any alternate thoughts on the control of the GPIO for bracketing the write then?

What exactly do you mean?

huntc, 2025-02-11:
> > Any alternate thoughts on the control of the GPIO for bracketing the write then?
> 
> What exactly do you mean?

I need to raise a GPIO representing my RS485 tx signal (push to talk). This needs to bracket my writes. Hence, was thinking of implementing my own writer for Port anyhow.

huntc, 2025-02-11:
> yes, you would actually need a `tokio-rs485`. Maybe you can use this project as a basis: https://github.com/mbr/rs485-rs

That's pretty old. :-) Also, I've previously used serialport-rs successfully to implement RS485 along with the GPIO behaviour based on timing. With MODBUS, the slave isn't permitted to start transmitting until the time it takes to tx 3.5 characters has expired, so I'm thinking it is ok to sleep for around 2ms having performed the write (assuming 19200 baud).

uklotzde, 2025-02-11:
> > yes, you would actually need a `tokio-rs485`. Maybe you can use this project as a basis: https://github.com/mbr/rs485-rs
> 
> That's pretty old. :-) Also, I've previously used serialport-rs successfully to implement RS485 along with the GPIO behaviour based on timing.

See also #229/#289.

huntc, 2025-02-11:
> See also [#229](https://github.com/slowtec/tokio-modbus/issues/229)/[#289](https://github.com/slowtec/tokio-modbus/pull/289).

Thanks. I did see those issues, but now I have my extra requirement around multiple slaves.

Here's what I'm thinking: a transport that implements AsyncRead and AsyncWrite where its implementations send to a channel using request/reply semantics (`oneshot` reply) and a separate task that receives on that channel and calls upon a user-supplied closure to do whatever is required e.g. call upon the existing client code. This would then support multiple slaves by virtue of their being multiple channel senders composed by a clone-able transport.

It would also support my own implementation of using `serialport-rs` directly and managing the tx GPIO I have.

Thoughts?

uklotzde, 2025-02-11:
>  ...and a separate task that receives on that channel and calls upon a user-supplied closure to do whatever is required e.g. call upon the existing client code.

Tasks should be spawn from application code, no need for callbacks and closures. The library should only provide the tools to implement such a task.

huntc, 2025-02-11:
> Tasks should be spawn from application code, no need for callbacks and closures. The library should only provide the ~tools~ API to implement such a task.

Agreed. WDYT of the idea in general though i.e. using a channel to back the AsyncRead/Write methods. I guess they'd be a bit of buffer copying happening, but that would be tolerable.

uklotzde, 2025-02-11:
Channels are also not needed if the application is able to drive the event loop and polls the bus.

huntc, 2025-02-11:
> Channels are also not needed if the application is able to drive the event loop and polls the bus.

Can you please elaborate... In summary, I want to attach multiple slaves. Thanks.

huntc, 2025-02-12:
I've now noticed that there's a `set_slave` method on the context... would a better approach be to always set the slave being targeted given a single context?

uklotzde, 2025-02-12:
> I've now noticed that there's a `set_slave` method on the context... would a better approach be to always set the slave being targeted given a single context?

That was the intention, switching between slave devices.

closed 2025-02-13 by huntc